FRC1076/RobotKitLib #9: Create the UltraSonic component interface for use in the robot.py framework.

mcolinj, 2020-10-31:
Look at the Freenove example code for a sonar unit, and move the function into a UltraSonic class so that it can be used.    Refer to the wpilib UltraSonic interface so we can keep it compatible.

EliThorpe, 2020-11-01:
Looks like the built in ultrasonic sensor class has three main functions:
`send_trigger_pulse()`
`wait_for_echo`
`get_distance`

To make a wpilib-compatable version, you will have to:
add a unit converter funciton, giving output in inches or mm
create a new counter program similar in function to the combination of `send_trigger_pulse` and `wait_for_echo`
and add the rest of the wpilib functions (should be pretty simple) to make it fully compatable

Might want to consider making a base sensor class, similar to the one wpilib has to make things easier when/if we get new sensors from electrical 

see Ultrasonic.py in the freenove code,  [wpilib documentation](https://robotpy.readthedocs.io/projects/wpilib/en/latest/wpilib/Ultrasonic.html), and [wpilib code](https://github.com/robotpy/robotpy-wpilib/blob/2019/wpilib/wpilib/ultrasonic.py

EliThorpe, 2020-11-01:
Then for the camera motor, you should probably put that in its own seperate class

mcolinj, 2020-11-02:
Camera motor should be an instance of the Servo class.      There is probably a reason to combine the two servos into something like a tilt/pan object.     But we don't have much control of the servos, and don't know *where* they are, so as an assembly, it would have limited capabilities.

mcolinj, 2020-11-05:
This job might look intimidating at first, but you can work on it in stages.

1.  Look at the Freenove code and see how the Ultrasonic interface works.
https://github.com/Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/tree/master/Code/Server

2.  Manually run it, maybe surround it with a loop that reports range values to convince yourself that it does work.   You don't have to have the tilt/pan assembled for this.   You can just plug in the sonar unit and set it on the robot platform or another holder.  Or if you just have a Raspberry Pi and a sonar unit, you can wire them up for testing.

3. Then think about how you could package it up so it looks more like the UltraSonic interface in wpilib.

Zoom420, 2020-12-02:
I hope someone else worked on this issue today, cause I sure didn't!

Zoom420, 2020-12-07:
https://github.com/Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/blob/master/Code/Server/Ultrasonic.py
The link to the existing sensor code from Freenove

Zoom420, 2020-12-07:
It looks like the Freenove code handles getting the distance by sending 3 trigger pulses in quick succession and taking the middle value, a clever internal data smoothing. We may or may not want to adopt this either as data smoothing or in addition to computing a rolling average/throwing out bad values.