FRC1076/RobotKitLib #12: Add a requirements.txt

numerator, 2020-11-01:
It would be good to add a `requirements.txt` with all python dependencies and instructions so that each dev can get their environment set up correctly (`pip/pip3 install -r requirements.txt`). Also might be good to add links in the readme to wpilib docs and any other docs that are necessary for understanding the code. For training it would also be good if API calls in the code were commented with where the associated docs are (assuming they're not all in the wpilib docs).

EliThorpe, 2020-11-01:
I think the requirements is already covered by issues #2 and #3. Documentation is definitely something we need more of

numerator, 2020-11-02:
Agreed. Sorry shoulda read the other issues before posting.

closed 2020-11-05 by EliThorpe