FRC1076/RobotKitLib #15: Experiment with and write up a design document for the line-following sensor.

mcolinj, 2020-11-02:
Describe how it works.   (Sensors return what kind of values, and what do they indicate?)

Describe how we could use it in our existing robot architecture.      In autonomous mode, could we hand off control to the drivetrain to this component, or would we just it in the main loop to just ask for corrections and then apply them to the drivetrain?

Propose a software interface for this line following sensor.

mcolinj, 2020-11-10:
Look at the line following code from the Freenove kit.    Try to tease out some information about what it is doing. 

https://github.com/Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/blob/master/Code/Server/Line_Tracking.py

If you are having trouble getting started, paste some bits of that code into here and add comments/question to it.      FYI, the line follower uses infrared sensors to detect dark vs. light regions.    I don't think it is very precise.    But there appear to be 3 of them.    Look at the sensor assembly and see if you can find the sensors.    Then think about what values they might return based upon the position of the line in relation to the sensors.

MineeCat, 2020-12-02:
Haven't started on this yet because I've been working on other issues (with the robot and vscode). Will work on it once I finish with those.