FRC1076/RobotKitLib #30: Add a laptop battery indicator to the driverstation GUI

EliThorpe, 2020-11-18:


KingDee241, 2020-11-18:
Need some extra info... where can I find the code for the driverstation?

mcolinj, 2020-11-18:
It is always good to ask for more information,

For the driverstation side, you will find the code:
          https://github.com/FRC1076/RobotKitLib/blob/main/driverstation.py
e.g. go to the repo, click on "Code", then on driverstation.py

You will want to look at pygame documentation for widgets, or maybe start by adding some button like thing based on Eli's code.     @EliThorpe does main have what we want for GUI now?

EliThorpe, 2020-11-23:
https://github.com/FRC1076/RobotKitLib/blob/main/driverstationgui.py

UI Code

KingDee241, 2020-11-23:
Trying to run the program on my computer... upon running driverstation.py, an error message appears, saying 
Traceback (most recent call last):
  File "d:\Desktop\FRC2020-21 PiHiSamurai\RobotKitLib\driverstation.py", line 4, in <module>
    import pygame
ModuleNotFoundError: No module named 'pygame'

It looks like something is wrong in the imports, and this is in the master branch. What should I do?

EliThorpe, 2020-11-23:
sounds like you need to install pygame

@KingDee241

run
`python pip install pygame`

mcolinj, 2020-11-24:
That little factoid belongs in the "How to get the robot driving" document.

KingDee241, 2020-11-27:
I've tried to run that command, but there is no indication that pygame was installed. Is there a specific location that I need to be in in order to install?

mcolinj, 2020-11-27:
You got a Mac or Windows machine?

matthew

EliThorpe, 2020-11-29:
[Driverstation UI](https://user-images.githubusercontent.com/18630012/100530578-bcbec480-31c1-11eb-87ad-15ac8f81e70a.png)
@KingDee241 The laptop battery indicator should go in one of the two open white spaces, either between the teleop button and voltage, or under the various indicators 

Have a look at some of these stack overflow pages for getting the computers battery:

https://stackoverflow.com/questions/16380394/getting-battery-capacity-in-windows-with-python
https://stackoverflow.com/questions/45626937/how-to-get-battery-percentage-with-python

KingDee241, 2020-12-02:
Progress update for 12-1-2020
none made. Still getting robot set up

EliThorpe, 2020-12-04:
Unassigning myself, this will be a good project for students to work on

closed 2021-01-11 by EliThorpe