FRC1076/RobotKitLib #33: Create the mecanum drivetrain

mcolinj, 2020-11-20:
We have 4 sets of mecanum wheels that could be used with the Freenove robot kit.    Design has to whip up some adapters so we can attach them to our gearboxes, but if you bring the software to life, you get a set of wheels.    The design project is described briefly in:   https://docs.google.com/document/d/1uuLvkiprWOmSIRJ5a0krY75xL-tG6-JNXKQO5vm4HaM/ which is on the google drive and entitled "Current Design Projects".

Could be fun, and not all that hard to do.  (getting the axle adapter is probably a much harder part)

matthew

mcolinj, 2020-11-20:
[image]

EliThorpe, 2020-11-23:
Have a look at https://github.com/robotpy/robotpy-wpilib/blob/2019/wpilib/wpilib/drive/differentialdrive.py

and https://github.com/FRC1076/RobotKitLib/blob/main/pikitlib/pikitlib/differentialdrive.py

mcolinj, 2020-11-23:
And then look at the mecanumdrive in wpilib

mcolinj, 2020-11-23:
Also take a look at some simple robot programs to see how the drivetrains are used.   Basically, they just take the values from the joystick, maybe a gyro (or joystick combined with sensor correction values), and takes those as input.   Then, they transform those values into the specific motor settings necessary to move the robot in the way intended by the joystick.

Here is an example of a arcade style drivetrain.    The important parts are creating it, and having it accept the driving parameters.

https://github.com/robotpy/examples/blob/master/arcade-drive/robot.py

And here is an example of a mecanum drivetrain.     Looks mightly similar.  (except for the fact that it needs to control 4 motors instead of only 2).

https://github.com/robotpy/examples/blob/master/mecanum-drive/robot.py

So the main task is transplanting the math code from wpilib in our rendition the library.  (in a way similar to the way that the arcade-drive(differential-drive) was transplanted.     To be successful, you really only need to implement two functions (the __init__()  and the driveCartesian() function.      Don't worry with anything related to motor safety, mostly because I'm not sure we've figured it out.  (that's setSafetyEnabled() and setExpiration())

IngmarNT, 2020-12-02:
I have been learning about drivetrains, the different kinds of input styles, and how I would port the current system to work with mecanum. I do not have any code currently (other than small stuff I've been messing around with).

Zoom420, 2020-12-09:
I have assigned myself because I was already working on the issue w/ Ingmar and figured I would make it official. I want to make some comfortable/intuitive control mappings w/ ingmar that will make more sense than the ones already in wpilib