FRC1076/RobotKitLib #35: Set up the pi so that run.py (robot) starts up automatically on reboot

mcolinj, 2020-11-21:
The robot really needs to start up automatically on reboot.  (disabled, of course).

The easiest way to do this is to do something like I've done in the past for the drivercam
software.     Have a shell script in the .bashrc that checks to see if the robot is running,
and starts it if it is not.

mcolinj, 2020-11-23:
Long ago I wrote some code to start up the stream drivercam server on the raspberry pi.    It is a script the runs when the user opens up a shell (which happens automatically on this pi when it boots up).     Anyway, see if you can make sense of the code.    Ask me questions about it.    Remember, it is checking to see if drivercam is running, we would need to check for a different name (maybe we need a better name than run.py?)

https://github.com/FRC1076/Vision_2017/tree/master/Core/2016_Fall_and_2017_Spring/drivercam

See what you think.      This contains a script which updates a few files on the pi.    Maybe we could deploy this to folks,  and ask them to run it.

matthew

primevaldragon, 2020-12-01:
if [[ $(pidof mjpg_streamer) ]];
then
    echo "mjpg_streamer is already streaming!";
    echo "View stream at: http://$HOSTNAME:$PORTNUM/?action=stream";
else
    echo "mjpg_streamer is *NOT YET* running, so we'll start it up!";
    cd /usr/src/mjpg-streamer/mjpg-streamer-experimental

I think I just need to change this to check for the program then launch it if its not running
I think I just need to change the cd and then the if

mcolinj, 2020-12-01:
That seems about right.    What you can do is test it out.    Create a program of your own, say: austin.py
that just loops, prints, sleeps, and see if you can write a .bashrc that will start it on bootup, as well as refuse to start it if it is already running.    If you have a monitor and keyboard, you can edit the .bashrc using VSCode, or you can use nano to edit it over ssh.     You can also check interactively if commands work before you run them in the script.      Think about how we might deploy this software to the current user base.

numerator, 2020-12-06:
Could also do this as a Linux service: https://www.hostinger.com/tutorials/manage-and-list-services-in-linux/#:~:text=Linux%20Services,as%20they%20lack%20an%20interface.&text=On%20the%20other%20hand%2C%20in,ends%20with%20the%20letter%20d.

This would restart the robot code whenever it goes down, not just on reboot.

primevaldragon, 2020-12-13:
I followed this video https://www.youtube.com/watch?v=cZa1oCSdbUk and it worked

EliThorpe, 2020-12-14:
Implemented with commit   fa4bc6a7ab342568f86b81f1f718fdfbb45b1d94

closed 2020-12-14 by EliThorpe