FRC1076/RobotKitLib #36: Create the documentation on how to bring the Freenove robot to life

mcolinj, 2020-11-22:
1.    Assemble the robot drivetrain and attach the raspberry pi.    Include any assembly advice and pointers to the documenation.    Also how to get to the documentation.

2.   Equip your driverstation.    (how to install all of the required software on the driverstation)

3.   How to bring up the robot.
      Cabling for monitor/keyboard, connecting to wifi, etc...

4.   How to find your robot.     how to find the robot on the network.   (many possible, many should be explained)

5.  How to start the robot running and the driverstation running.

6.  How to update the robot software

numerator, 2020-11-30:
The documentation lives here: https://docs.google.com/document/d/13n1QN8vziR7Z6QPZxJy76q95Cz_2N0AkUMWgDU8ddN0/edit?usp=sharing

Still a living document as people encounter new problems, so keeping this issue open for now.

EliThorpe, 2020-12-02:
Should we move this to the Readme?

numerator, 2020-12-02:
Maybe leave it on Google for now until everyone has gotten through the instructions. Easier for group editing and commenting on Google (esp. while people don't know GitHub). We can pickle it and put it into the README once we're sure it's stable.

EliThorpe, 2020-12-02:
I meant we could close the issue and put the link to the instructions in the readme

numerator, 2020-12-02:
Ah. Then yes. That would work!

numerator, 2020-12-02:
Just realized though: the document is currently world-writable. Perhaps
should be changed to allow comments but not editing. I'll do that.

numerator, 2020-12-02:
Fixed.