FRC1076/RobotKitLib #40: robot code is slipping

MrRSquared, 2020-11-25:
This new run.py is excellent. Our car is running well. However, there is an error in the console on the pi.
It seems whenever we run a drivetrain object (in autonomousPeriodic or teleopPeriodic), we get slipping in the timing. 
`WARNING:root:Auton has slipped by 64.0 miliseconds!`
Interestingly, this does not seem to happen when we control the motor directly (through the speedController interface) using code .set(.5) or a joystick.
It does happen when we control the Leds using the Led class that FreeNove wrote (but that class has a lot more to it than the drivetrain class you all wrote).

EliThorpe, 2020-11-26:
Having had a look at your light code, there are a lot of looping code, way to much to be directly in teleop periodic. It would probably be best to move the lights code to a separate thread

MrRSquared, 2020-11-26:
Hello,
Yes. This is odd. I just tried it again with the main repo in this git as is (no changes to robot.py). I have both auto and tele slipping. I wonder what may be the cause. I will play with some debugging and report back if I uncover anything.

As for the light code, I agree. Actually, the code up there is old. Now, the only loop in the light code that is running is their rcolorWipe() method, which I just realized does implement a sleep function, so naturally, that will create a slip.). But, I do encounter the error irrespective of the light code.

MrRSquared, 2020-11-26:
Yes. When I took the sleep function out of the Led code, that worked without slipping. I could not figure anything else out thus far however.

MrRSquared, 2020-11-29:
Thank you for looking into this. I am closing this issue as I think I did not have a requirement installed. after whipping my install disk and beginning with a fresh install (using the desktop version os), there is no slippage at all in teleop or auto.

closed 2020-11-29 by MrRSquared