FRC1076/RobotKitLib #45: Clean up analog input class

EliThorpe, 2020-11-29:
Needs to be rewritten to be cleaner and better coded, also needs more comments

mcolinj, 2020-12-10:
The analog input is interesting because it is designed to support two different ADC chips.   You can see in the init that it sends a code on the i2c bus to discover which chip is on the bus.   I think they may have changed chips from an earlier version of the board and they want the software to support either.      At any rate, they store the chip identity in a string and then switch on it so they can use a different function later.    I think you know how to do that better.    Also, the code could use some comments describing what it is doing.   Also, we could figure out which chip is on the version 2.2 board that we are using and decide to eliminate the unneeded code.    (it could be put into a different AnalogInput implementation that could, perhaps be used by someone who had the older board.

mcolinj, 2020-12-10:
Once we know which chip they are using, we could focusing on making that code better.    They use a different approach to getting a value.     See if you can figure out how they differ.       It may be a challenge, but I think you are up to it.