FRC1076/RobotKitLib #61: Error establishing connection

MrRSquared, 2020-12-10:
Hello,
If this is still known and taking you all down the wrong road, or if it is too much of an interruption (perhaps too early?), please ignore/ close it. However, I am still unable to connect to the robot since the introduction of the deployment process which could be a bug (either in documentation or software). 

Here is what I am seeing on my end with a fresh install of RobotKitLib...

After install of git bash and RobotKitLib on the host machine, if on the robot, I start 
`$ python robot_setup.py`
I get it waiting for the connection. If I just launch driverstation.py on the host machine, It will not connect with no error other than "No Comms"
However, if I launch `deploy.py <IP address>` in git bash on the host machine, I get the following error.
```
Traceback (most recent call last):
  File "deploy.py", line 35, in <module>
    p = subprocess.Popen("./create_archive.sh", stdout=subprocess.PIPE)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38\lib\subprocess.py",
 line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\user\AppData\Local\Programs\Python\Python38\lib\subprocess.py",
 line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 193] %1 is not a valid Win32 application
```
Again, I do not mind being patient, but if this were a bug, I would want to know.

Thank you again for this incredible project. I am enjoying it immensely, and I believe it will help us out a lot.

EliThorpe, 2020-12-10:
This issue sounds like you are not using a bash terminal, I suggest git bash (download here https://git-scm.com/downloads). Since the error seems to be that create_archive.sh "is not a valid Win32 application", which is true, since its a bash script.

MrRSquared, 2020-12-10:
That does sound like what is going on. I will investigate a solution. I am using git bash, but somehow my environment is not set up correctly. When launching deploy.py `py deploy.py <IP address>`, the python script runs in bash, but the bash script does not seem to. I will investigate. If I figure it out, I will post a solution and close this.

MrRSquared, 2020-12-10:
I did make some progress. It seems this is a known issue. Windows is... [fussy](https://stackoverflow.com/questions/58402900/how-to-run-bash-commands-using-subprocess-run-on-windows). I edited deploy.py to send the file by doing this around line 35.
```
if os.name == 'nt': #NT fix courtesy of https://stackoverflow.com/questions/58402900/how-to-run-bash-commands-using-subprocess-run-on-windows
        try:
            p = subprocess.Popen(['C:\Program Files\Git\\bin\\bash.exe','-c',"./create_archive.sh"], stdout=subprocess.PIPE)
        except:
            raise
```
It now sends the file, but I may not be sending the correct file because on the robot I see...
```
Got a connection from  (<IP>)
hash type:  abc
file name:  RobotKitLib.tar.gz
file size:  1013760
File received successfully.
tar: RobotKitLib.tar.gz: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
rm: Connection closed.
```
Interesting note, the file size increases each time I run deploy.py.

commit 5065bd88 (2020-12-11): fix #61
closed 2020-12-11 by EliThorpe